ut-issl/c2a-core #7: CMakeLists.txt を何とかする

meltingrabbit, 2021-11-18:
## 概要
CMakeLists.txt を何とかする

## 詳細
全体的になんとかしたい．  
非推奨表記などはなくしたい．

## close条件
全体的に見直しが完了したら．

## 備考
NA

sksat, 2021-12-15:
`.c`なファイルをC++としてビルドしたいことがある，Visual Studioでもビルドできてほしいといった制約のため，非推奨表記の撲滅には至らなかったが，だいぶなんとかはなった(メンテ可能になった)．

sksat, 2022-06-15:
あ〜だいぶなんとかしたので閉じてよさそう。どういうかんじになんとかしたか箇条書きで雑に書いて閉じるかんじにします。

sksat, 2022-07-05:
- 文脈の整理
  - C2A の `CMakeLists.txt` は元々 SILS(S2E + C2A) でのビルド用
  - これは Windows(MSVC) 想定
  - S2E の `CMakeLists.txt` から `add_subdirectory()` されることで使われていた
  - 最近は SILS 専用というわけでもない(ex: https://github.com/arkedge/c2a-user-for-raspi)
- 基本方針
  - S2E と C2A の `CMakeLists.txt` をちゃんと分離する(不要・非自明な依存関係の排除・C2A 単体ビルドサポート)
  - メンテナンス性の向上
  - 複数環境のサポート
- 主ななんとかしたポイント
  - CI で複数環境でのビルドを行い，Windows(MSVC) 以外の環境・コンパイラのサポート
  - (SILS用の)C++としてではなく，C89 としてビルドするオプションを追加(このCIも追加)
  - include path などの S2E から刺し込まれていた C2A のビルド設定を C2A 側に移行
  - 共通に使用するファイルパスを CMake の変数として切り出した(ex: `C2A_CORE_DIR`)
  - 共通のビルドオプションなどを `common.cmake` に切り出した
  - グローバルに影響するビルド設定を可能な限り排除し，各ビルドダーゲット毎に設定するようにした
  - ディレクトリ毎に `CMakeLists.txt` を分割した
  - example minimum user を(CMakeビルド環境向けの) exampleとしても位置付け，こちらも同様に改善
  - これに伴い，C2A core/user のビルド設定の分離も実施