ut-issl/c2a-core #30: ビルドCI Actions の job nameが同じで，branch protectionで区別できない

meltingrabbit, 2021-11-24:
## 概要
ビルドCI Actions の job nameが同じで，branch protectionで区別できない

## 詳細
どっちも build_minimum_user になってるので．

https://github.com/ut-issl/c2a-core/blob/62e835382a108c6ec688c798d4b918e8ad28b598/.github/workflows/build_as_c89.yml#L18-L19

## close条件
名前をユニークにしたら

sksat, 2021-11-24:
これworkflowじゃなくてjobの名前なのかなり非自明ですね

meltingrabbit, 2021-11-24:
それな～～～～ workflow nameにしてほしすぎる．

sksat, 2021-11-24:
まあjobを区別できるのは便利かもしれないですが，しかしなあ．
やるにしても"workflow.job"みたいなかんじになっててほしさある．GitHubにfeedback送るべき案件かもしれない．

meltingrabbit, 2021-11-24:
https://github.com/github/feedback/discussions これか？

sksat, 2021-11-24:
ここもある
https://support.github.com/contact/feedback

meltingrabbit, 2021-11-24:
送った．

meltingrabbit, 2021-11-24:
https://github.com/ut-issl/c2a-core/pull/31 のとおり，なんと branch protection rule で設定したものが全部パスしないとマージできないとかいうシロモノだった．

なので，最後のcommitが指定したCIすべてを回すもので，かつすべて通らないと，マージできなくなってしまう．．．

meltingrabbit, 2021-11-24:
とはいえ，一旦job nameのコンフリクトは解消されたのでcloseする．

closed 2021-11-24 by meltingrabbit