ut-issl/c2a-core #33: documentsを入れる

meltingrabbit, 2021-11-25:
## 概要
documentsを入れる

## 詳細
C2Aとは，各種tips，パラメタの説明，コーディングルールなどをいれる．

どこにどのように入れるかも議論する．

## close条件
いれたら

meltingrabbit, 2021-11-25:
- c2a-documents といったリポジトリを切ってもいい？
- /docs を切っていれる？
  - 様々な user sample が今後できるが，どのリポジトリにいれる？

meltingrabbit, 2021-11-25:
C2Aなのであまりないかもしれないが，画像を入れると，リポジトリ全体の容量が肥大化するなどもある？（無視できるか？）

また，ある程度圧縮した画像をドキュメントに差し込むことになるとは思うが，その後の更新・編集作業のためには，編集生データも保存しておきたかったりするかもしれない？

sksat, 2021-11-25:
画像データというかバイナリデータは基本的にGitで管理すべきではないのでなあ．容量デカくなるのもそう(とはいえ，それを気にしすぎてもしょうがない)．

sksat, 2021-11-25:
あんまり`./docs`がデカくなってきたら分離するのでよさそう

meltingrabbit, 2021-11-25:
C2Aはまあいいかぁ．S2Eは検証系のグラフがたくさんあって，すぐ肥大化しそうな気がしている．

sksat, 2021-11-25:
たしかに．まああと容量がデカくて困るのは普段使うブランチがデカくて`git clone`が重くなるのがメインだと思うので，orphanブランチを切ってそこをGitHub Pagesにするのもアリかもしれない．

meltingrabbit, 2021-11-25:
想像以上にテクいな < orphan ブランチ

docsはpagesいらない気がするのよね．所詮mdだし．コードの詳細はreferenceがあるから，込み入ったことは書かないはず．  
なので，個人的には /docs ででかくなったら考える，でいい気がする．

orphan ブランチはなるほど，使ってみたい，といった気持ちがあるので，S2Eとかでやってみたいね．

sksat, 2021-11-25:
orphanブランチでやるの完全にGitHub Pagesあるあるですね．まあそもそも最終生成物をcommitすること自体が渋いんですが...

meltingrabbit, 2021-11-25:
> まあそもそも最終生成物をcommitすること自体が渋いんですが...

そうねぇ．pagesだと，成果物pushするしかないし，staticなページしか無理だからねぇ．

sksat, 2021-11-25:
static site，なんだかんだそんな困らないですよ．むしろ動的コンテンツを提供したいことは(この用途的にも)ほぼ無い(のでstatic siteはそこそこ流行ってる)．
それはそれとして最終生成物をcommitしないといけないGitHub Pagesは渋くて，なので個人的にはGitHub Pages以外のstatic site hosting service推しです(VercelとかCloudflare Pagesとか)．

meltingrabbit, 2021-11-29:
規約だけでも先にコミットしてしまうか？

meltingrabbit, 2021-12-04:
入れた．

closed 2021-12-04 by meltingrabbit

meltingrabbit, 2021-12-19:
別での議論を，備忘録としてここにも残す．

> なので個人的にはGitHub Pages以外のstatic site hosting service推しです(VercelとかCloudflare Pagesとか)．

ISSLとして管理するものが増えるのは大変（もうすぐ卒業の鈴本が管理したくない）というのもあって，ISSL orgで静的サイトをだすならgithub pages一択，で．

sksat, 2021-12-19:
補足: GitHub Pages，ようやく最終生成物をcommit&pushしなくてもよくなるかもしれない雰囲気も醸し出して来たので，その点でも意外とGitHub Pagesのままでよいかもしれませんね
https://github.blog/changelog/2021-12-16-github-pages-using-github-actions-for-builds-and-deployments-for-public-repositories/
> However, in the future this will enable us to give you the ability to fully customize your pages build and deployment workflow to use any static site generator you want without having to push the build output to a special branch of the repository.

meltingrabbit, 2021-12-19:
> build and deployment workflow

期待じゃん．