ut-issl/c2a-core #34: testをいれる

meltingrabbit, 2021-11-25:
## 概要
testをいれる

## 詳細
pytestで書かれたやつ．

どこにどのように入れるかも議論する．

## close条件
いれたら

meltingrabbit, 2021-11-25:
https://github.com/ut-issl/c2a-core/issues/33 を議論していて思ったけれども，testも同じリポジトリ（userならそのuser-repositry/test，coreのcore部分なら c2a-core/test/）でいいのでは？と思ってきた．

今までは，リポジトリを分けていた（hoge_mobc_pytest リポジトリがあった）けれど，不要？

meltingrabbit, 2021-11-25:
を議論していて思ったけれども，testも同じリポジトリ（userならそのuser-repositry/test/，coreのcore部分なら c2a-core/test/）でいいのでは？と思ってきた．

今までは，リポジトリを分けていた（hoge_mobc_pytest リポジトリがあった）けれど，不要？

sksat, 2021-11-25:
テストはむしろ本体にくっついていないと微妙まである

meltingrabbit, 2021-11-25:
CIつくってて，びみょいなと実感してるところだった．（CI側でバージョンの一致が辛くなってた）

meltingrabbit, 2021-11-25:
> coreのcore部分なら c2a-core/test/

これはだめだな．coreのパラメタはuser側で設定されるので，core開発用のcoreのテストはcoreのminimum user に入れる必要があるな．

meltingrabbit, 2021-11-27:
テストCIは https://github.com/ut-issl/c2a-core/issues/16 で引き継ぐ

closed 2021-11-27 by meltingrabbit