ut-issl/c2a-core #36: src_user/IfWrapper/Sils/*.c にC++のコードを置くことについて議論する

meltingrabbit, 2021-11-25:
## 概要
src_user/IfWrapper/Sils/*.c にC++のコードを置くことについて議論する

## 詳細
- https://github.com/ut-issl/c2a-core/pull/35 でいろいろ話が出た．

## close条件
なんとなく今後の方向性が見えてきたら．

sksat, 2021-11-25:
流石に分かりにくいのでC++っぽい拡張子にしましょう．S2Eと合わせるなら`.cpp`か．

sksat, 2021-11-25:
https://github.com/ut-issl/c2a-core/pull/35#issuecomment-979321172
ファイル名が変わるのでS2Eも困るからタスク切った方がいいな，と思ったけど，これはよく考えたらS2Eの`CMakeLists.txt`が **すごいことになってる** ことによるものであって，向こうの`CMakeLists.txt`も直したら一瞬で解決する問題だった．

sksat, 2021-11-25:
区別したいという元の気持ちからするとC++なヘッダは`.hpp`とかにしたい気もするが，S2Eのヘッダは普通に`.h`なのよな

meltingrabbit, 2021-11-25:
> .cpp, .hpp

これはそうんだね．そうすべきだとおもう．

sksat, 2021-11-25:
まあS2Eと一貫性が無くなるだけ(どうせこれをincludeするのもS2Eだけ)なので，区別できるようにするのを優先して`.hpp`にしてもいい気がするな．

sksat, 2021-11-25:
> [#35 (comment)](https://github.com/ut-issl/c2a-core/pull/35#issuecomment-979321172) ファイル名が変わるのでS2Eも困るからタスク切った方がいいな，と思ったけど，これはよく考えたらS2Eの`CMakeLists.txt`が **すごいことになってる** ことによるものであって，向こうの`CMakeLists.txt`も直したら一瞬で解決する問題だった．

よく考えたらこれはなんも困らなかった．include pathの指定が渋いだけなので．

sksat, 2021-11-25:
置くことについて議論する，というより既にそうなっていて非常にわかりにくい上に #35 で実用上の問題も発生している，といった状態なのでタイトル変えるか

PR #37 (2021-11-28): [minimum_user] SILS用のラッパー部分のソースファイルを.cppにする
## 概要
Fix #36 

SILS専用のラッパー部分のC++としてしかコンパイルされない`.c`なファイルの拡張子を`.cpp`にする．

## Issue
- #36 

## 詳細
N/A

## 検証結果
CIが通り，かつS2Eとリンクできればよし

## 影響範囲
S2Eが変更するファイル名に依存していればS2E側も修正する必要があるが，ヘッダを変えなければほぼ無いだろうと思われる．

## 補足
N/A
closed 2021-11-28 by sksat