ut-issl/c2a-core #51: CCSDS周りのコードをcoreで管理する?

meltingrabbit, 2021-11-27:
## 概要
ひとことで

## 詳細
詳しく

## close条件
どうなったらcloseできるか

## 備考
なにかあれば

## 注意
- 関連する Projects が存在する場合，それの紐付けを行うこと
- 可能ならば `priority` ラベルを付けること
- 可能ならば Assignees を設定すること
- close するときは結論を明記すること

meltingrabbit, 2021-11-27:
## CCSDS周りの整理

今，結構ぐちゃってるので（だいぶまともにはなってきたが），core移植を前に，整理したい．

鈴本がCCSDSわからんので，通信系のレビューがほしい．

CCSDS周りはこれをみながら書いている:   
- https://sma.jaxa.jp/TechDoc/Docs/JAXA-JERG-2-401.pdf
- https://sma.jaxa.jp/TechDoc/Docs/JAXA-JERG-2-402.pdf

### 現 TCP

tlm cmd packetという意味でC2A初期から使われてそう．

C2Aの抽象化パケットCTCP (common tlm cmd packet) の実体．

space packetと同義なので， space_packet.c/h 構造体名も SpacePacket に変更したい．  
接頭辞は SP_．

データ長や，2nd headerはユーザー定義だが，ISSL標準として，策定したい．パラメタは，
- 2ndヘッダに8byte絶対時刻を含めるか含めないか
- 最大パケット長（CCSDSとして使うものは432だろうけれど，非MOBCはメモリ効率から小さくしたいのでユーザー定義に）

テレメもコマンドもこれを使っていいんですよね？（2ndヘッダは異なるが．）

### 現 M_PDU

テレメトリ．

Multiplexing Protocol Data Unit？

であれば， multiplexing_protocol_data_unit.c/h 構造体名も MultiplexingProtocolDataUnit にしたい．  
接頭辞は M_PDU．

### 現 TcpToMPdu

発破解体したい．が，優先度低

### 現 VCDU

テレメトリ．

virtual_channel_data_ U_unit.c/h にし，構造体名もVirtualChannelDataUnit にしたい．  
接頭辞は VCDU_．

これの定義は， https://sma.jaxa.jp/TechDoc/Docs/JAXA-JERG-2-402.pdf にはないが，何を見ればい？  
これだと， M_PDU は VCDUではなく， AOS TransferFrameに入っているように見える

### 現 TC Segment

コマンド．

https://sma.jaxa.jp/TechDoc/Docs/JAXA-JERG-2-401.pdf だと，ただの segment

TCつけとく？

だとすると，

tc_segment.c/h にし，構造体名も TcSegment としたい．  
接頭辞は TCS_．

### 現 TC Frame

コマンド． TCはテレコマではなく，テレコマンド？（TC，ややこしい．．．）

tc_transfer_frame.c/h にし，構造体名も TcTransferFrame としたい．  
接頭辞は TCTF_．

meltingrabbit, 2021-11-27:
ISSLのあつかう複数のOBCにとって，この機能はcoreだが，OSSとしては，CCSDSの機能はそこまでcoreではないというのが難しいところ．CCSDS非準拠の衛星もおおい．

meltingrabbit, 2021-12-07:
@yngyu @yanagida-kanta これ，以下の理由から進めたいんですが，やっぱ微妙ですかね？

- C2A間通信のアップデートをかけていきたい
- https://github.com/ut-issl/c2a-core/issues/64 などの改修をuser部でやるのはしんどい

なので，coreにいれつつ，使わない人はビルド対象から外してもらう，という運用で，どうでしょう？

core実装の仮定としては，

- CTCPは以下のようにユーザー定義してもらって，デフォルトはTCPをつかう
  - https://github.com/ut-issl/c2a-core/blob/62e835382a108c6ec688c798d4b918e8ad28b598/Examples/minimum_user_for_s2e/src/src_user/Settings/CmdTlm/common_tlm_cmd_packet_define.h#L11 
- CCSDS配下は，coreではincludeしない

meltingrabbit, 2021-12-07:
ついでに，exec typeなどで macro cmdの表記が残ってるので，BLCになおしたい．

yngyu, 2021-12-07:
common_tlm_cmd_packet なのに TCP がデフォルトなのは違和感がありますが、coreに移すのは良いと思います

meltingrabbit, 2021-12-07:
> common_tlm_cmd_packet なのに TCP がデフォルトなのは違和感があります

あれ？これなんで？（TCPはSpacePacketにrenameされます）

yngyu, 2021-12-07:
なんか語感的に common って付いてる方がデフォルトっぽい... って思ったからです。良くわかってなかったんですが TCP と CTCP の使い分けってなんでしたっけ...? なんかそこら辺の定義ちゃんとしたい、みたいな issue があったのは覚えてるんですが

yngyu, 2021-12-07:
あれでも common って付いてるってことは TCP の中にCTCP とか specific TCP とかあるって感じなんですかね... すいません言っててわからなくなってきました

meltingrabbit, 2021-12-07:
CTCPはCCSDS関係なく，C2A内部でとびかうcmd, tlmの構造体定義．

で，TCPはCCSDSで規定された，コマンドテレメ定義．

んで，今は，C2A内部のテレコマもTCPでいいじゃんって使ってる感じ．でも2nd OBCにとっては特にTCP使う必要もないし，CCSDSの規定はユーザー定義領域もあるので，TCP自体はuser依存（現在もMOBCとAOBCで違う定義）．

CTCPは，そういったTCPとか，その他のパケット定義を抽象化したもので，C2AではこのCTCPというものに対してデータ操作を行う，って感じ．

user側，具体的には，

https://github.com/ut-issl/c2a-core/blob/develop/Examples/minimum_user_for_s2e/src/src_user/CmdTlm/common_tlm_cmd_packet.c

でTCPをCTCPとしてアクセスするIFを定義してあげてる（つまりこれはuser依存．CTCPとしてたとえばHogePacketを使うなら，ここにHogePacketへのアクセスを書く）

yngyu, 2021-12-07:
なるほど、理解しました。ありがとうございます

meltingrabbit, 2021-12-08:
手始めに，一旦TCPだけcore移植するか．その仮定でexec typeなども整理しよう． https://github.com/ut-issl/c2a-core/issues/64 もあるし．

とはいえ，TCPのAPIDなどはどうしてもuser依存なので，そのあたりの切り出しはやらねばならん．ヘッダファイルのみならず，関数実装（TCPとCTCPのexec type変換とか）もあるので，.c側もuser/core分離せねばならんなぁ．

meltingrabbit, 2022-01-21:
ひとまずTCPは移動しよう

https://github.com/ut-issl/c2a-core/issues/205

meltingrabbit, 2023-11-08:
https://github.com/arkedge/c2a-core/issues/196 で続きはやる

closed 2023-11-08 by meltingrabbit