ut-issl/c2a-core #64: UTCなどの絶対時刻でTLC (Time Line Command) が打てるようにする

meltingrabbit, 2021-11-29:
## 概要
UTCなどの絶対時刻でTLC (Time Line Command) が打てるようにする

## 詳細
- OBC_TL.Cmd_HOGE において，時刻引数をtiではなく，utcなどの絶対時刻にて打てるようにする．
  - 既存の OBC_TL.Cmd_HOGE は残すので，あらたに追加することになる．
    - 命名は？ OBC_ATL (absolute tl)？，OBC_UTL (utc tl)？
- これは，あくまでエイリアス的な機能であることに注意する
  - 現在， OBC_TL.Cmd_HOGE は TCP_CMD_EXEC_TYPE_TL のTCP_EXEC_TYPEとなり，それはCCP_EXEC_TYPE_TL0のCCP_EXEC_TYPEに対応し，TL0に展開される．
  - ここで作るものは，コマンド受信部 　(https://github.com/ut-issl/c2a-core/blob/31dbe6e8c347252b55b6fc63ef7c11016b75e9b5/CmdTlm/packet_handler.c#L77) では，別のTCP_EXEC_TYPEとして受理されるが，最終的にはCCP_EXEC_TYPE_TL0に変換されるべき（つまり， https://github.com/ut-issl/c2a-core/blob/31dbe6e8c347252b55b6fc63ef7c11016b75e9b5/CmdTlm/packet_handler.c#L94 にルーティングされるべき）である．
  - つまり，引数はUTCなどであるが，それがPacketHandlerにて，TIへ変換され（これはObcTimeの関数で行う）た後に，TL0コマンドに変換され，TL0のキューに登録される．
- それに応じて，TCP_CMD_EXEC_TYPE はこのCmd用のtypeを追加する必要がある．
  - TCP_CMD_EXEC_TYPE などは 別途資料 を参照
- その他の注意
  - 新しい TCP_CMD_EXEC_TYPE の対応や，コマンドファイル対応などでWINGSの改修が必要．
    - これは早めにWINGSと調整すべき．C2Aの実装が終わる前にもWINGS側はリリースして問題ないはず．はやくリリースされたほうが，SILSなどでの検証が楽．
  - ObcTime (https://github.com/ut-issl/c2a-core/blob/31dbe6e8c347252b55b6fc63ef7c11016b75e9b5/System/TimeManager/obc_time.h) を一度整理してもいいかもしれない
    - 結構汚いコードが残ってるので
    - 型とか公開関数とか整理しても良いかも

## close条件
実装ができたら

## 備考
- https://github.com/ut-issl/c2a-core/issues/63 の娘issue

meltingrabbit, 2021-12-07:
そう言えば，うるう秒～～

isseu化した

- https://github.com/ut-issl/c2a-core/issues/84

meltingrabbit, 2021-12-27:
@hirotaka912   cc @chutaro 

UTLのWINGS側，ひとまず
- `TCP_CMD_EXEC_TYPE_UTL` は `0x*4`
- TCP headerのTime Indicatorは，`2020-1-1-00:00:00UTC を epoch （基準）`として，10倍したもの（0.1秒分解能．`(time - utl_epoch) x 10` ）でお願い．

chutaro, 2022-01-28:
これですが、「UTL cmd を打つ」という点において C2A の実装は終わっていて、wingsの0.1秒精度の改修後に変更が必要なのは pytest だけなので、close しちゃっていいですか？改修対応は以下で。

- https://github.com/ut-issl/c2a-core/issues/175

meltingrabbit, 2022-01-28:
OKです！

closed 2022-01-28 by chutaro