ut-issl/c2a-core #80: c89のgcc Wextra Warningに対応すると，あるマイコンIDEにてWarningが出てしまう

meltingrabbit, 2021-12-05:
## 概要
c89のgcc Wextra Warningに対応すると，あるマイコンIDEにてWarningが出てしまう

## 詳細
- https://github.com/ut-issl/c2a-core/pull/73 にて， `const uint32_t` などの返り値を `uint32_t` などにした
  - `warning: type qualifiers ignored on function return type [-Wignored-qualifiers]` が発生 
  - 基本型をconstつけて返しても，値渡しなため意味がないから 
- 一方で，以下などでは，const構造体のメンバをconstなしで返している，として，あるマイコンIDEでは `C0008 (W) Conversion in return` を出してしまう．
  -  https://github.com/ut-issl/c2a-core/blob/1280836a5c6a6a5fc7a4fc8a5a9211134345b5b5/Drivers/Super/driver_super.c#L1513-L1516

## close条件
どちらの環境でもWarningがでなくなったら

## 備考
どうしようか．とりあえず，上の例だと，意味はないんだけど `(uint32_t)` とキャストさせる？

sksat, 2021-12-06:
これ某IDEの方のwarningはコンパイラの実装が素朴なことによって出ているので微妙なんですよねえ．まあキャストかなあ．

sksat, 2021-12-06:
あとGCCの方のwarningは組み込み型だからとかではなく，型修飾子である`const`を関数の戻り値に指定してもしょうがないからです．右辺値にconst付けてもしょうがない，といった方がいいかな．

meltingrabbit, 2021-12-06:
> 右辺値にconst付けてもしょうがない

そうだね．ディスクリプション雑すぎたので直しておきます 🙏

meltingrabbit, 2021-12-06:
> 某IDE

Warningメッセージでググると特定されるな笑

まあ，学会などでマイコン情報出してるので，問題ないと言えば問題はない．

meltingrabbit, 2021-12-06:
これ，ちょっと気持ち悪いのが，

`const DS_StreamConfig* p_stream_config` なので，`p_stream_config->tx_frame_size_` の値そのものは const なので，これキャストすると，const キャストになっちゃうんだよね．

meltingrabbit, 2021-12-16:
まだでてる．

- [ ] common_tlm_cmd_packet_util
- [ ] command_analyze

meltingrabbit, 2021-12-16:
一旦きえた？

ひとまず様子見

meltingrabbit, 2021-12-22:
また出たらissueをきるとして，close

closed 2021-12-22 by meltingrabbit