ut-issl/c2a-core #124: OBCT，TMGRの細々した改善

meltingrabbit, 2021-12-21:
## 概要
OBCT，TMGRの細々した改善

## 詳細
- doxygenコメントのない連中のコメント追記
- testの追加（ないものは）
- `time_manager.c` でコマンド関数はファイル末にまとめたい
- 規約修正
  - https://github.com/ut-issl/c2a-core/pull/116#discussion_r772514713
  - `master_clock_` , `OBCT_unixtime_info_` の変数名
  - というか，これら， `TimeManager` 構造体つくってまとめてもいいのでは？ と言う気がしてる（そっちのほうがC2Aっぽい & UTL 実装時に情報（時刻originとか）をここでもてる？）
- Cmd引数の取得方法を https://github.com/ut-issl/c2a-core/pull/70 に準拠 

## close条件
終わったら

## 備考
- https://github.com/ut-issl/c2a-core/pull/116 からの派生
- からなずしも１つのPRでやる必要はない

@chutaro 頼んだ！

chutaro, 2021-12-22:
> `TimeManager`構造体つくってまとめてもいいのでは？

これは、`ObcTime`構造体と`OBCT_UnixtimeInfo`構造体を`TimeManager`のメンバーとして持たせるという意味ですか？それとも解体してTimeManagerにまとめるという意味ですか？

meltingrabbit, 2021-12-22:
`ObcTime`構造体と`OBCT_UnixtimeInfo`構造体の定義は引き続き obct.hで持つのでいいと思う．これは汎用データ型なので，ここで定義すべき．

で，tmgr.hの
https://github.com/ut-issl/c2a-core/blob/94e228155c9f964cf3a5199c5b010f0371c48cd2/System/TimeManager/time_manager.h#L7-L11
を

```
typedef struct
{
  ObcTime master_clock_;
  OBCT_UnixtimeInfo unixtime_info_;
  utl_unixtime_origin_;   // UTL cmd実装時にここに追加
  struct
  {
    ObcTime initializing_time;
    uint8_t initializing_flag;
  } init_info_;
  
} TimeManager;
```

な感じにすればいいのでは？とおもってる

chutaro, 2021-12-22:
なるほどです！

meltingrabbit, 2021-12-22:
@chutaro https://github.com/ut-issl/c2a-core/pull/130 がマージされたので，そこで終わったものは，issue descriptionのチェックボックス，更新しておいて～

chutaro, 2022-01-14:
TMGR に関してのテストは追加し終わったので Closeする。UTL_cmdに関するテストは以下の別issueで
- https://github.com/ut-issl/c2a-core/issues/175

closed 2022-01-14 by chutaro