ut-issl/c2a-core #129: GSCD，TLCD，RTCDなどの実行時エラー履歴を残す

meltingrabbit, 2021-12-22:
## 概要
GSCD，TLCD，RTCDなどの実行時エラー履歴を残す

## 詳細
これらの実行時エラーがCDIS構造体でしか管理されていないせいで，過去のエラーを追いずらい．  
record event でよさそう？

## close条件
記録できるようになったら．

chutaro, 2022-02-22:
方針を相談したいのですが、（返信は全然急いでないです）

- CDISを呼んでいるインスタンスはcoreだと以下の5つですが、
  - GSCD
  - RTCD
  - TLCD
  - task dispatcher
  - DI_AOBC
- エラー記録方法はどちらが良いでしょうか？前者の方がどこでエラーが出たか瞬時に分かるので良い気はするのですが。。
　1. インスタンスごとに別々のgroupをEL_CORE_GROUPに作成し、local=CCP_EXEC_STS
　2. 実行時エラーは全て group=CDIS_EXEC_STS とかにまとめて、local=CCP_EXEC_STS、note=インスタンスの種類enum

- 前者の場合、CDIS_dispatch_command の返り値を void から CCP_EXEC_STS に変更するのが一番楽かなと思うのですが、変更してしまっていいですか？

meltingrabbit, 2022-02-28:
あ～，これ確かに，どうするんだ．

> note=インスタンスの種類enum

これ，どうやて CDISの中に入れようとしてる？（区別しようとしてる？）

meltingrabbit, 2022-02-28:
https://github.com/ut-issl/c2a-core/blob/05f6c07751413dcc3654e02924ff71442fc2a4f1/TlmCmd/command_dispatcher.c#L87 

という感じなので， cdis ポインタアドレスをnoteに入れてしまえば，インスタンスは区別できる？？

あと，別件で

https://github.com/ut-issl/c2a-core/blob/05f6c07751413dcc3654e02924ff71442fc2a4f1/TlmCmd/command_dispatcher.c#L16-L26

というenumは追加してしまった（よしなに変更しても消してもいいよ）

@chutaro

chutaro, 2022-03-01:
command_dispatcher.c 内に
> typedef enum
> {
>CDIS_TYPE_GS,
>   CDIS_TYPE_RT,
>   CDIS_TYPE_TL,
>   CDIS_TYPE_TDSP,
>   CDIS_TYPE_DI_AOBC
> } CDIS_TYPE

のようなenumを作って、CommandDispatcher構造体にもメンバーとして持たせて、

> CDIS_init(CDIS_TYPE_RT, &PH_rt_cmd_list)

のように初期化すれば、`CDIS_dispatch_command`内部で CDIS_TYPE ごとに場合分けして、
group=EL_CORE_GROUP_TLCD_EXEC_STS, local=CCP_EXEC_STS, note=cmd_code みたいにできますね。

chutaro, 2022-03-01:
エラーが起きたインスタンスとCCP_EXEC_STSを記録すれば十分なのであれば、

> cdis ポインタアドレスをnoteに入れてしまえば

これで行きたいと思います。cmd_codeも記録したかったり、groupの時点で発生場所を区別したかったりする場合は、上記のコメントみたいな実装かなー、と思うのですが。
ELにどこまで記録するかみたいな加減が正直あまり分かっていないのですが、どうなんでしょうか？

chutaro, 2022-03-01:
これですが、CIDSはCDISのミスですか？
https://github.com/ut-issl/c2a-core/blob/05f6c07751413dcc3654e02924ff71442fc2a4f1/TlmCmd/command_dispatcher.c#L16-L26

meltingrabbit, 2022-03-01:
せっかく CDISは各実行インスタンスと独立できてるので，enumつくるのはさける方向で行きたいね～（管理がめんどくさい）

meltingrabbit, 2022-03-01:
> これのCIDSはCDISのミスですか？

ミスっぽいので，直してもらえると 🙏

closed 2022-04-01 by chutaro