ut-issl/c2a-core #138: GSから直接insert, deleteできるTL0を2つ（以上）に分割したい

meltingrabbit, 2021-12-27:
## 概要
GSから直接insert, deleteできるTL0を2つ（以上）に分割したい

## 詳細
- Bus用のTLとMission用のTLを分離したい
- Misson側でTLを専有したり，delete allしたりできる
- とはいえ，メモリ使用量も増え，TaskListの占有率もあがる

## close条件
分割検討して，分割したら

meltingrabbit, 2022-01-21:
@chutaro 追加するTL (MISSION TL, MTL？) は，無効化できるように作ってください． `#define` とかで．

例えば `EL_IS_ENABLE_MIDDLE_ERROR_LEVEL` のように．（ELのMIDDLE_ERROR_LEVELはユーザー選択）

chutaro, 2022-01-21:
了解です。自分の理解の確認ですが、
TL0：GS用 ==> TL（バス用）とMTL（ミッション用）に分ける
TL1：BC用
TL2：テレメ用
ということですよね。

> static void TLCD0_init_(void);
static void TLCD0_dispatch_(void);
static void TLCD1_init_(void);
static void TLCD1_dispatch_(void);
static void TLCD2_init_(void);
static void TLCD2_dispatch_(void);

このあたりも番号命名ではなく、TLCD, TLCD_mis, TLCD_bl, TLCD_tlm みたいに変更しますか？

meltingrabbit, 2022-01-21:
です！

> TLCD, TLCD_mis, TLCD_bl, TLCD_tlm みたいに変更しますか？

そうねぇ．そうしますか．

chutaro, 2022-01-21:
create_app_info で 引数取れたらいいんですが。。

meltingrabbit, 2022-01-21:
それねぇ．．．

https://github.com/ut-issl/c2a-core/issues/126#issuecomment-1004506415

meltingrabbit, 2022-01-21:
TLCDのためだけにそれすんの？？ってなっちゃって，それはないな～～ってなってる．

chutaro, 2022-01-21:
了解です。とりあえず TL3 をミッション用に増やして動くこと確認して、必要であればそのあと命名変更する方針で行きます！

meltingrabbit, 2022-01-21:
https://github.com/ut-issl/c2a-core/blob/ae105ab79dd1121e0c0bcc0c855318c21c9b05ef/Applications/timeline_command_dispatcher.h#L9-L15

が

```
typedef enum
{
  TL_ID_REGISTERD_FROM_GS_FOR_BUS = 0,
  TL_ID_DEPLOY_BC        = 1,
  TL_ID_DEPLOY_TLM     = 2,
#ifdef TL_IS_ENABLE_MISSION_TL
  TL_ID_REGISTERD_FROM_GS_FOR_MISSION = 3,
#endif
  TL_ID_MAX
} TL_ID;
```
みたいになるんだと思ってます．

とはいえ `TL_ID_REGISTERD_FROM_GS` 長すぎなのでなんとかしたいｗ

meltingrabbit, 2022-01-21:
`TL_ID_FROM_GS` , `TL_ID_FROM_GS_FOR_MISSION` とかでもいいかもな

chutaro, 2022-01-21:
ありがとうございます、これでやってみます。

chutaro, 2022-04-06:
分割したのでclose

closed 2022-04-06 by chutaro