ut-issl/c2a-core #162: 改行コードについて整理する

meltingrabbit, 2022-01-04:
## 概要
コーディング規約に

https://github.com/ut-issl/c2a-core/blob/94a4007362ae5192bfead701672391a0323e1a79/Docs/General/CodingRule.md#L135

とあるが，現状これはWindows環境でcloneすると auto crlfがかかってるだけなので，repository上ではLFになっている気がする．

機械的に統一/チェックをする，もしくはコーディング規約を変えるのか，といったことを整理したい．

## 詳細
- 現状のコーディング規約ではCR+LFに統一することを要求している
- この規約は機能/強制しておらず，現状のリモートリポジトリのソースコードのファイルの改行コードはLFになっている
- CR+LFは特定の開発環境(Renesas)から来ている制約であり，本当に要求されるのはここでビルドする際にCR+LFになっていることのみ
- Gitの設定で`core.autocrlf=true`になっていれば作業時にはCR+LFになる
- Git for Windowsのデフォルト設定が`core.autocrlf=true`であるため，なんとかなってはいる
- 改行コードを何にするにしても，これは各自の設定任せではなく`.gitattributes`で設定すべき

## close条件
整理できたら

sksat, 2022-01-04:
現状もリモートリポジトリ全部LFっぽいですし，規約の方を変えて`.gitattributes`追加だけでよさそう？

meltingrabbit, 2022-01-04:
その説が濃厚だと思ってる．

meltingrabbit, 2022-01-04:
これちゃっちゃとやっちゃうか．next core update欄に移動した．

meltingrabbit, 2022-01-06:
適当に関係者と話して，この方向で行くことにした