ut-issl/c2a-core #169: どのバージョンのツールを使っていいかわからなくなるので，適合バージョン情報をreadmeにいれる

meltingrabbit, 2022-01-06:
## 概要
どのバージョンのツールを使っていいかわからなくなるので，適合バージョン情報をreadmeにいれる

## 詳細
TlmCmdDBやCodeGeneratorなどのtoolの適合バージョンをreadmeにいれる．  
これによって，あるバージョンのcoreを使うuserにとって，利用しやすくなる．

## close条件
入れたら

## 備考
本当はなにかしらの依存関係スキーマをつかいたいが．．．

meltingrabbit, 2022-01-06:
いろいろ @sksat と話をして，tool側で最小c2a coreバージョンを明記するのが良さそうとなった．

今鈴本が考えているのは，各toolのmainを更新したら
- releaseを切る
- そこにそれに合わせた c2a-core側のPRのリンクを貼る（一緒にそれのマージされたときのdevのhashもはる？）
をする．

toolが更新されたらc2a側のバージョンを上げるかどうかはちょっと要検討（patchぐらい上げてもいいのか，とはいえmain以外でバージョン上げるの，ちょっとフローが，などあり）

sksat, 2022-01-06:
toolが更新されるような時は確実に非互換な更新なので，それを含む更新がmainにマージされる時はminorを上げるようにする(規約にする)，ぐらいでいい気がしています．

meltingrabbit, 2022-01-06:
> それを含む更新がmainにマージされる時はminorを上げるようにする(規約にする)，ぐらいでいい気がしています．

それはそうしましょう．

https://github.com/ut-issl/c2a-core/issues/141 とあわせて規約整備しますわ．

meltingrabbit, 2022-01-17:
各Toolsに現状の対応とreleaseを打った．

- https://github.com/ut-issl/c2a-enum-loader/releases
- https://github.com/ut-issl/c2a-tlm-cmd-code-generator/releases
- https://github.com/ut-issl/python-wings-interface/releases
- https://github.com/ut-issl/tlm-cmd-db/releases

closeする．

cc @200km 以後，このようにtools側のrelease noteで適合バージョンを明示していきます．

closed 2022-01-17 by meltingrabbit