ut-issl/c2a-core #170: バージョン情報の記載のあるheaderを作る

meltingrabbit, 2022-01-06:
## 概要
バージョン情報の記載のあるheaderを作る

## 詳細
```
#define C2A_CORE_VER_MAJOR 3
#define C2A_CORE_VER_MINOR 4
#define C2A_CORE_VER_PATCH 0
```

みたいな

## close条件
できたら

## 備考
議論点は

- どのフォルダに配置する？
- いつどのように更新する？

sksat, 2022-01-06:
場所的には `c2a_core_main.h`では，と思っています(何にも属していないのでルートにあるのが望ましい & ルート直下のファイルを増やしたくない)

meltingrabbit, 2022-01-06:
バージョンしか書かれてないヘッダのほうが，toolなどで読んだときとか，自動で値更新するとき（があるかわからんが）に便利そうって思ったりはした．（あと，適当な場所で使うときにincludeするときとかもわかりやすい．バージョン番号つかうのだけに `c2a_core_main.h` をincludeしたくないなと．）

meltingrabbit, 2022-01-06:
> ルート直下のファイルを増やしたくない

はそうなので，一旦それで，で，今後困ったらファイル分割するか～

sksat, 2022-01-06:
toolで読む時，自動更新は別にどこにあってもそんなに変わらないかな，と思っています．
ただ，バージョン番号使うためだけに `c2a_core_main.h`をincludeしたくない，はかなりあると思っていて，個人的にはファイル分けたいですね．ルート直下にファイル増えることを許容できるか次第．

meltingrabbit, 2022-01-06:
> バージョン番号使うためだけに

実際，これはほぼなさそう．commit hashはtlmにいれたり，デバッグ表示にいれたりするので使うけど

sksat, 2022-01-06:
そうなんですよね．そんなにプログラム中で使う気はしないというのはある．

meltingrabbit, 2022-01-06:
じゃあ一旦 `c2a_core_main.h` で．

あとはどう更新するかか．main update prでもいいが，そうするとmain mergeではじめて上がった記載ファイルをどうdevまで持ってくるかが課題．

これ，devをmain merge後に毎回rebaseするのってどうなんだ？（まあいいのかな）

meltingrabbit, 2022-01-06:
dev rebase って，PRでできるのか，というのも気になり

sksat, 2022-01-06:
> dev rebase って，PRでできるのか，というのも気になり

これ何を気にしてます？

meltingrabbit, 2022-01-06:
手元でのforce pushにならない？って．（devとmainはいまforce push禁止してる

meltingrabbit, 2022-01-06:
あー，ff mergeのpr出せばいいのか？

meltingrabbit, 2022-01-06:
といってもいま，repository設定で--no-ffマージしか許されていない

sksat, 2022-01-06:
そうだった... > force push/--no-ff

sksat, 2022-01-06:
じゃあmain update PRでやればいいのでは

meltingrabbit, 2022-01-06:
> じゃあmain update PRでやればいいのでは

あーーー，頭がついてなかった．

そっか，main update prってdevにcommitできるじゃん．

sksat, 2022-01-06:
実はそう

meltingrabbit, 2022-01-06:
あれ，devはpush禁止してるので，commitできない説

sksat, 2022-01-06:
アッ

meltingrabbit, 2022-01-06:
- devのバージョンあげますPR（すぐmain updateするよ `. |ω・)ﾁﾗ`
- main update PR

の2段？？．．．

sksat, 2022-01-06:
dev push解禁しない限りそれしかないかあ

meltingrabbit, 2022-01-06:
厳密にはtagよりも古いcommitにバージョン含まれちゃうが，まあ動作に影響のあるファイルには更新ないので，いいか．

meltingrabbit, 2022-01-06:
ではそれで行くかぁ．

https://github.com/ut-issl/c2a-core/issues/169 とあわせて，リリースフロー，ドキュメント化します．

sksat, 2022-01-06:
> dev push解禁はないなぁ

ですよねー．adminだけpush可みたいなかんじにできればまた別なんですが．

meltingrabbit, 2022-01-06:
まあ，それもそれで事故こわいんで，一旦これで運用しよう，でクソだるかったらadmin解禁含めて考えよう．

sksat, 2022-01-06:
人間ではなくリリース担当Botを生やしてそいつにやらせる，などはないではない(が，運用コストを考えるとノーチャンな気がする)

meltingrabbit, 2022-01-17:
https://github.com/ut-issl/c2a-core/blob/develop/Docs/General/release.md でやった．

更新については，やってみてだるかったらまた考える．

close．

closed 2022-01-17 by meltingrabbit