ut-issl/c2a-core #175: UTL cmd のテスト対応

chutaro, 2022-01-12:
## 概要
UTL cmd のテストを追加する

## 詳細
- [ ] python-wings-if で UTL cmd を打てるようにする
- [ ] testの追加

## close条件
上記が完了したら。

## 備考
なにかあれば

meltingrabbit, 2022-01-19:
## テストの残り
- https://github.com/ut-issl/c2a-core/pull/182 の残
  - https://github.com/ut-issl/c2a-core/blob/69b265402fbf19f84aa7affd663a2a929afac857/Examples/minimum_user_for_s2e/src/src_user/Test/test/src_core/System/TimeManager/test_time_manager.py#L220
  - https://github.com/ut-issl/c2a-core/blob/69b265402fbf19f84aa7affd663a2a929afac857/Examples/minimum_user_for_s2e/src/src_user/Test/test/src_core/System/TimeManager/test_time_manager.py#L112-L121
    - ここは，10個も登録する必要はないし，TL Tlmを見るより，Cmd登録時に `PH_TLC_PAST_TIME` が出てることを確認すべき

chutaro, 2022-05-16:
完了したのでclose

closed 2022-05-16 by chutaro