ut-issl/c2a-core #183: 衛星内部のマスター時刻であるTIと絶対時刻であるunixtimeの変換の精度を向上させる補正を入れる

meltingrabbit, 2022-01-14:
## 概要
衛星内部のマスター時刻であるTIと絶対時刻であるunixtimeの変換の精度を向上させる補正を入れる

## 詳細
- https://github.com/ut-issl/c2a-core/issues/181 などに関連
- `OBCT_STEP_IN_MSEC` , `OBCT_CYCLES_PER_SEC` などは，あくまでざっくりとした値であり，「TIが1増えると絶対時刻が100msふえる」という対応は，厳密ではない
- したがって，これを補正する項を設定できるようにし，この時刻変換精度を上げる．

## close条件
できたら

chutaro, 2022-01-21:
思ったことメモ（ひとりごとです）

- `OBCT_STEP_IN_MSEC = 1 `と保証されない場合、 `get_*_in_msec` 系の関数の返り値が整数でなくdouble型になるので、`get_*_in_sec` と本質的にやってることが一緒になり、存在意義がない。管理も面倒なので全て消してしまいたい。
- 同様に、`OBCT_sec2cycle`なども引数や返り値を double に変更する？
- CYCLE の意味が以下の二つで異なる。前者は正確に0.1秒なのに対し、後者はOBCの時刻精度に依存。ちょっと分かりにくい。
  - UTLコマンドの実行引数は epoch起算の CYCLE(=0.1秒) 刻みの整数値
  - MOBC の CYCLE は0.1秒
- utl_unixtime → TI に変換するとき、現状では unixtime_at_ti0 を utl_unixtime に変換して引き算をしていたが、時刻補正が反映されないので以下の3ステップを踏む必要あり
  - utl_unixtime を unixtime に変換
  - diff = unixtime_at_ti0 - unixtime
  - diff を TI に変換

meltingrabbit, 2022-01-21:
`OBCT_sec2cycle` とか， `get_*_in_msec` とかは，補正入れなくていいよ（そこまで精度のいい数字求めてないので）

とりわけ精度のいるUTL周りと，unixtime <> TI 変換まわりだけ，補正を入れてほしい

chutaro, 2022-01-21:
わかりました。`get_*_in_msec` を使う場面は全部`get_*_in_sec`で代替できそうだから必要ないなーとか思ったのですが、とりあえず補正の部分だけ実装します。

meltingrabbit, 2022-01-21:
基本的に，C2A内部（BCとかは，TLとか）では，誤差みたいな細かいこと考えずにシンプルでやってるからね．

https://github.com/ut-issl/c2a-core/blob/ae105ab79dd1121e0c0bcc0c855318c21c9b05ef/Examples/minimum_user_for_s2e/src/src_user/Settings/Modes/Transitions/InitialSL.c#L15
とかでdouble 帰ってこられても困っちゃうのよね．

chutaro, 2022-01-21:
なるほど、ユーザー側で結構つかわれているんですね

chutaro, 2022-02-17:
実装終わっているのでクローズ。user側での補正項デフォルト値の決定は以下で引き継ぐ。

- https://gitlab.com/ut_issl/issl6u/cdh/operation/-/issues/142

closed 2022-02-17 by chutaro