ut-issl/c2a-core #202: PH_add 系の返り値がバラバラ

yngyu, 2022-01-21:
## 概要
PH_add_ 系の返り値がバラバラ なので統一したい

## 詳細
https://github.com/ut-issl/c2a-core/blob/ae105ab79dd1121e0c0bcc0c855318c21c9b05ef/TlmCmd/packet_handler.c#L84
PH_add ~ 系の関数が返り値が成功した場合
* PH_add_gs_cmd_: `PH_SUCCESS`
* PH_add_tl_cmd_: `PH_TLC_REGISTERD`
* PH_analyze_block_cmd_: `PH_BC_REGISTERED`
* PH_add_rt_cmd_: `PH_REGISTERED`
* PH_add_utl_cmd_: `PH_TLC_REGISTERD`
と多様性に溢れている。全部 PH_SUCCESS に揃えたい。また、PH_analyze_block_cmd_ もadd_block_cmd_ に揃えたい。

## close条件
直したら

meltingrabbit, 2022-01-21:
ごめん，間違えた．．．．（close押しちゃったのでreopenした）

meltingrabbit, 2022-02-06:
> と多様性に溢れている。

これ，実はさ，多様性に溢れてるので， gs_ph_ackをみると，打ったCmd種別（GSCなのかTLCなのかとか）が分かって便利だった，って説あったりする？

とはいえ，`PH_SUCCESS `に揃えるべきなので，代わりにgs infoに，最後に受け取ったCmd種別（CCP_DEST_TYPE と CCP_EXEC_TYPE）を保存しておいても良いかもね．

yngyu, 2022-02-06:
確かにそうですね、追加しておきます

yngyu, 2022-02-06:
これ一つ悩んでいることがあってPHでのエラーなので `PH_BC` とか `PH_TL` とかを無くしたいんですがそうすると BC とか TL の時点で掴んだエラーはどうするべきなんだというのがあり...。C2A階層が上に行くにつれて下のエラーが握りつぶされて行って残ってない印象があるんですよね

meltingrabbit, 2022-02-06:
なくしたいのはそうなんだよなぁ．．．今，PHとBC，TLが密結合（ph.cのなかにBCとかTLの関数があるので，PH_ACKにそれがあるのは自然）だから，そうなってる感じよね．

ちょっとコード見るのでおまち

meltingrabbit, 2022-02-06:
`PH_ACK` 系， `PH_SUCCESS` → `PH_ACK_SUCCESS` にしたいね（かぶるので）

meltingrabbit, 2022-02-06:
`PH_add_tl_cmd_` と `PH_analyze_block_cmd_` の返り値どうします？って問題よね．．．

PHがrouting関数であって，routing先がC2A coreの機能，って考えたら，まあ `PH_BC_REGISTERED` でも良いのでは？って気はしてきたなぁ．．．

というのも， `PH_user_cmd_router` が `PH_FORWARDED` を返すのも悪くない気がするのよね．（逆にここで `PH_SUCCESS ` を返すのはなんか微妙）

meltingrabbit, 2022-02-06:
それか，`PH_SUCCESS` は `PH_SUCCESS` を返す（このC2AのPH的にエラーはなかったって意味で）んだけど，別の情報として，何をしたか（BC登録したよ，とか），なんでエラーでたか（BC IDがおかしかったよ，とか）を持つのはありなのかね？（関数返り値，2個返せないので，ちょっとだるい？（PH_ACKとERR_ENUMの2個のstruct返す？））

meltingrabbit, 2022-02-06:
あとは，細かいエラーはEvent loggerにしちゃうってのもありかもな

yngyu, 2022-02-06:
細かいエラーは EL 記録は大いにありだと思っています。ただソースコードがエラーハンドリングまみれになるのがちょっと嫌だな、って思います。

meltingrabbit, 2022-02-06:
たしかにねぇ．．．

一旦，今使ってない `PH_INVALID_CRC` とか消してみて，あとは絶対おかしいだろ，みたいなのも変えてみて，考えてみる？（今PH_ACK，明らかにおかしい状態なので）

yngyu, 2022-02-06:
https://github.com/ut-issl/c2a-core/tree/feature/modify_ph_ret  
ここで今一旦変えてみたバージョンがあって今手元でテストで通してみています

meltingrabbit, 2022-02-06:
あ～～～～，いや，これさぁ， `PH_analyze_cmd_packet` ってGSからCmd打った時だけじゃなくって，App内部でCmd発行したいときも使うので，やっぱエラー情報が呼び出し元まで戻らないと困る気がしてきた．（つまり，BC IDおかしかったよ，とかも返したい）

meltingrabbit, 2022-02-06:
diff見たいので，PR出してもらっていい？

yngyu, 2022-02-06:
出しました

yngyu, 2022-02-10:
https://github.com/ut-issl/c2a-core/pull/242  
マージしたので close します

closed 2022-02-10 by yngyu