ut-issl/c2a-core #211: CTCPのCCPとTCP分離にともなう PacketList の対応

meltingrabbit, 2022-01-24:
## 概要
`CTCP` の `CCP` と `TCP` 分離にともなう `PacketList` の対応

## 詳細
- https://github.com/ut-issl/c2a-core/issues/204 の一環
- https://github.com/ut-issl/c2a-core/pull/210 でまとめて実装を諦めて，分離したもの
- PL を，CTCP でも CCP でも TCP でも使えるようにする

## close条件
なんとかなったら

## 備考
- これができないと， https://github.com/ut-issl/c2a-core/issues/206 ができない．

yngyu, 2022-01-24:
* packet list は packet list ではなく連結リストとしてデータ構造にしたい

meltingrabbit, 2022-01-24:
わかる．とはいえテンプレートのないCでは，なかなかタフだよねぇ．

本来はデータ構造とデータとアルゴリズムは直行させるべきなので，牛くんのそれは正しいんだけど，この状況だと，ひとまずCTCP, CCP, CTPに限定的に対応する片方向リストを実装するしかないのかもなぁ．．．と思ったり（もうちょっと考えます）

yngyu, 2022-01-24:
```c
struct
{
  void* node;
  struct NodeTag* next;
} NodeTag;
typedef struct NodeTag Node;
```
みたいに定義する
初期化の時点で
```c
CCP command_list_ [LIST_SIZE];
Node node_list_ [LIST_SIZE];
for (i = 0;i < LIST_SIZE; ++i)
{
  node_list_[i] = &command_list_[i];
}
```
で初期化して現 PL_initialize に node_size = sizeof(CCP); で引き渡す.  
insert の copy の時には packet_list に size情報が既にわたっているのでそれを元にコピーする.  
でどうでしょうか

meltingrabbit, 2022-01-24:
```
struct
{
  void* node;
  struct NodeTag* next;
} NodeTag;
```

これは確定でいいと思う．データを制限する場合でもしない場合でも，こうするべきだよね．

meltingrabbit, 2022-01-24:
それ， `node_list_[].node` が片方向リストの次のpacketではなく自身のpacket指してない？

もし，次のpacketを指した場合は，今度は逆に `command_list_` と `node_list_` の対応が取れなくなりそう．

yngyu, 2022-01-24:
次のパケットは next で指していて、node ポインタは自分の実体を指してます

meltingrabbit, 2022-01-24:
あー，そうか．

meltingrabbit, 2022-01-24:
行ける気がしてきた？

で，packetのサイズ渡しておけば，packetのcopyはできるので，insertやdeleteはできるっていう．

yngyu, 2022-01-24:
問題は node ポインタが void* なのでNULLポは避けられてもsize違いによるスタックなんとかかんとかが起こる可能性を0には出来ないって点ですね...

meltingrabbit, 2022-01-24:
今，主にdeploy bcの高速化で，packetのcopyをそのままcopyするんじゃなくて， `CTCP_copy_packet` をつかうことで，有効長さだけcopyし，高速化してるよね．この技は使えなくなるけど，そもそもCTCPのCCP，CTP分離でCCPは小さくなるから，いいのか．

meltingrabbit, 2022-01-24:
> size違いによるスタックなんとかかんとか

ってなんだｗ？

meltingrabbit, 2022-01-24:
> そもそもCTCPのCCP，CTP分離でCCPは小さくなるから，いいのか．

いや？ 再プロなど考えて，めっちゃでかいcmd paramを取れる設計にしてたら，詰むなぁ．．．TL deployの3-3出ちゃいそう

yngyu, 2022-01-24:
Tlm list に Cmd_node 渡して Tlm長分コピーしようとして、でもCmd分のサイズしかなくてコピーが行き過ぎるやつです、あのマクロでも考えられた奴です

meltingrabbit, 2022-01-24:
あー，あれね．それはがんばってアサーションしっかりするしかないね．

meltingrabbit, 2022-01-24:
Cだからアサーションできないじゃん・・・・・・（initで渡されたpacket情報，voidだからサイズわかんねぇ．．．

yngyu, 2022-01-24:
それなんですよ、

yngyu, 2022-01-24:
int* で + 1 すると4byte飛んでくれるやつ、なんかうまく使えないんですかね、なんであいつサイズわかってるんだろう

meltingrabbit, 2022-01-24:
Utilつくって，普段はそれ経由でinitかなぁ．．．（CCP用，CTP用のinitをつくる）

yngyu, 2022-01-24:
って思ったんですが void* として受け取った瞬間に全部意味無い、おしまい

meltingrabbit, 2022-01-24:
> int* で + 1 すると4byte飛んでくれるやつ、なんかうまく使えないんですかね、なんであいつサイズわかってるんだろう

コンパイル時型情報だから．
voidにしたら無理．

yngyu, 2022-01-24:
Nodeにsize持たせれば assertion 出来るんですが、安全取るかメモリ減らすか、ですね...

meltingrabbit, 2022-01-24:
> Nodeにsize持たせれば assertion 出来るんですが

それ入れる予定．だけど，sizeは初期化時に人間が与えることになるので，与え間違えると死

meltingrabbit, 2022-01-24:
C2Aの今後の未来は，低リソース側にあるから，メモリ使用量は減らす方向かなぁ．

yngyu, 2022-01-24:
了解です

yngyu, 2022-01-24:
> それ入れる予定．だけど，sizeは初期化時に人間が与えることになるので，与え間違えると死  
これはまぁそれのinit用の関数作れば回避はしやすくなりますね

meltingrabbit, 2022-01-24:
そうね．つまり

https://github.com/ut-issl/c2a-core/issues/211#issuecomment-1020243850

meltingrabbit, 2022-01-26:
ひとまず終わったので，close

closed 2022-01-26 by meltingrabbit