ut-issl/c2a-core #247: 新packetと新C2A間通信に対応した GENERATE_TLM Cmdを検討し実装する

meltingrabbit, 2022-02-08:
## 概要
新packetと新C2A間通信に対応した GENERATE_TLM Cmdを検討し実装する

## 詳細
- 現在のはこれ
  - https://github.com/ut-issl/c2a-core/blob/bfced5758a0d0e555d00399d51a471d7e478ce1b/TlmCmd/telemetry_generator.c#L15-L74
  - https://github.com/ut-issl/c2a-core/pull/217 でcategoryがなくなったので，一時的な対応をしている
  - また，DR記録とテレメ配送が同時にできるように，設定している
- 今後，これをどうしていくか考える．ポイントは
  -  新しいテレメトリフォーマットはこれ
    - https://github.com/ut-issl/c2a-core/blob/develop/Docs/Core/communication.md#secondary-header-telemetry
  - 既存の `Cmd_GENERATE_TLM` は後方互換性のために残す？
  - Cmd headerをすべて手打ちできる低レベルコマンドはつくる
    - `Cmd_GT_GENERATE_TLM` みたいな名前？
  - Destination Flags を打たなくていい便利コマンドを作る？
    - `Cmd_GT_GENERATE_HK_TLM`, `Cmd_GT_GENERATE_MS_TLM`, `Cmd_GT_GENERATE_DR_TLM`
    - DRだけ引数に DR Partition を追加
    - GENERATEって命名はどうなんだ？
- 別OBCからのテレメトリパケットを取得し流すコマンドも作る
  - `Cmd_GT_GENERATE_TLM_OF_OHTER_OBC` ?
  - 命名考えたい．．．キャッシュ or bufferから吸い出しているイメージの命名
  - ボードを指定するための引数は，APID？enum？
  - これも，無印とHK, MS, DR を作る？

## close条件
検討して実装したら

yngyu, 2022-02-08:
* 既存の Cmd_GENERATE_TLM は後方互換性のために残す？
  * 暫く残す(deprecated) で良いんじゃないでしょうか、その後の使い勝手次第で残すか消すか決めて良いと思います。
* Tlm headerをすべて手打ちできる低レベルコマンドはつくる
  * Tlm Header 手打ちってどこまで指定しますか? Header とか 2nd header まで含めますか?
* Destination Flags を打たなくていい便利コマンドを作る？
  * これは flag 複数あるので無しでいい気はします(今後ここらへんが変わると個別Generate_Tlmを変更しなければいけないのが面倒そう)
* 別OBCからのテレメトリパケットを取得し流すコマンドも作る
  * MOBCだけ特別で、他は各コンポにコンポID割り振ってGENERATE_COMPO_TLM で良い気がします、というかコンポIDではなくAPIDで良い?

meltingrabbit, 2022-02-08:
> Tlm headerをすべて手打ちできる低レベルコマンドはつくる

書き方悪かったね．`Dest Flags` と `DR Partition` です．

これ直打ちするモチベは，flagを複数あげて，DR記録しながらMS tlmとして下ろす，とかができるって感じ．

とはいえ，ほぼ使わない（使わず済むならつかいたくない）って感じなので，別途 dest flagごとのgenerate tlmコマンドほしいなぁって（そっちのほうがミスが少ない & コマンドファイル見たときに意図がわかりやすい）

meltingrabbit, 2022-02-08:
> 既存の Cmd_GENERATE_TLM は後方互換性のために残す

これは残しましょうか

meltingrabbit, 2022-02-08:
> MOBCだけ特別で、他は各コンポにコンポID割り振ってGENERATE_COMPO_TLM で良い気がします、というかコンポIDではなくAPIDで良い?

"コンポ" ってのが不明瞭な単語で，space pacektを出力しないコンポは，APIDはMOBCになるし，MOBCの普通のテレメと一緒なので普通に（新しく作るコマンドの方の） `Cmd_GENERATE_TLM` で大丈夫．

一方，space pacektを出力するAOBCとかは，パケットをMOBCでためていて，それを配送するためのコマンドが必要って意味かな．この場合，tlm idはMOBCとは独立なので，別途コマンドが必要．

meltingrabbit, 2022-02-08:
DRとの親和性考えて（DRからの再送，と別OBC packetのbufferからの再送って意味で）

```
Cmd_TG_REPLAY_TLM ${obc指定enum} ${tlm id} ${dest_flags} ${dr_partiton} ${送出回数}
```

ってのは？

APID，複雑だしAPIDから特定のOBCを引っ張ってくるの今の構造だと大変なので（Cだとdict型がないので．．．），enumかなぁ．．．と

yngyu, 2022-02-08:
> 書き方悪かったね．Dest Flags と DR Partition です．
これ直打ちするモチベは，flagを複数あげて，DR記録しながらMS tlmとして下ろす，とかができるって感じ．
とはいえ，ほぼ使わない（使わず済むならつかいたくない）って感じなので，別途 dest flagごとのgenerate tlmコマンドほし いなぁって（そっちのほうがミスが少ない & コマンドファイル見たときに意図がわかりやすい）思ったんだけど，なくてもいいか．
そしたら
Cmd_GT_GENERATE_TLM ${tlm id} ${dest_flags} ${dr_partiton} ${送出回数}
って感じかしら？

あると便利ではあると思うんですが将来的にCmdが増えた時に2^n で種類が増えると嫌だなぁ... と思いました。

yngyu, 2022-02-08:
> APID，複雑だしAPIDから特定のOBCを引っ張ってくるの今の構造だと大変なので（Cだとdict型がないので．．．），enumかなぁ．．．と
Enum作ってsubOBC叩くのとAPID経由でsubOBC叩くのって変わらなくないですか?APID持ってる = C2A 載っててTLM降ろせるでいい気がします

yngyu, 2022-02-08:
> Cmd_TG_REPLAY_TLM ${obc指定enum} ${tlm id} ${dest_flags} ${dr_partiton} ${送出回数}

これで良いと思いますが、個人的にはちょっと性質が変わるのでMOBCのDRと subOBCからのbufferTLMは分けたい気がします(結局内部でDR叩くか、buffer 叩くかで変わってきて、内部関数変わるので)

meltingrabbit, 2022-02-08:
> Enum作ってsubOBC叩くのとAPID経由でsubOBC叩くのって変わらなくないですか?

うん，変わんない．

enumだと，配列にできて，sub obcのtlm packet buffer叩くところをいい感じにできるなって思ったのと，APID長いなぁ．．．って思っただけなんだけど，新しいindex作るのはそれはそれで管理がめんどくさいし，今後の拡張考えたらAPIDにすべきなので，APIDにしますか．

> これで良いと思いますが、個人的にはちょっと性質が変わるのでMOBCのDRと subOBCからのbufferTLMは分けたい気がします(結局内部でDR叩くか、buffer 叩くかで変わってきて、内部関数変わるので)

変える予定よ．

DRはどのOBC packet入ってるかわからないし．（＝DRはAPIDをしらない）

DRは引き続き，

```
Cmd_DR_REPLAY_TLM
```

の予定

meltingrabbit, 2022-02-08:
tlm managerとの兼ね合いがちょっとめんどくさいなこれ．

`Cmd_TG_REPLAY_TLM` でAPIDが自分だったら `Cmd_GENERATE_TLM` に読み替えてほしいけど，それはそれでミスリーディングっていう．．．

meltingrabbit, 2022-02-10:
```
CCP_EXEC_STS Cmd_TLM_MGR_REGISTER_HK_TLM_FOR_GENERATE_TLM(const CommonCmdPacket* packet);
CCP_EXEC_STS Cmd_TLM_MGR_REGISTER_HK_TLM_FOR_REPPLAY_TLM(const CommonCmdPacket* packet);
CCP_EXEC_STS Cmd_TLM_MGR_REGISTER_HK_TLM_FOR_REPLAY_DATA_RECORDER(const CommonCmdPacket* packet);
```
みたいにするしかないのか？

meltingrabbit, 2022-07-27:
- https://github.com/ut-issl/c2a-core/issues/155#issuecomment-1027098869 はこっちで引き取る．

meltingrabbit, 2022-12-14:
- https://github.com/ut-issl/c2a-core/pull/474#issuecomment-1350215365 などの話もある

meltingrabbit, 2023-03-13:
口頭で話したメモ

```
Cmd_GENERATE_TLM      : これまでまま
Cmd_TG_GENERATE_TLM   : 自分のOBCテレメ生成
  引数: tlm id
      : Dest type (for DR 保存，など)
      : Dest info
      : 生成回数
Cmd_TG_REPEAT(REPLAY)_TLM  : 2nd OBCのテレメバッファー再送
  引数: APID or APIDを示すenum？
      : tlm id
      : Dest type (for DR 保存，など)
      : Dest info
      : 生成回数
Cmd_DR_REPLAY_TLM: これまで（categoryなどは TG_GENERATE に準拠）
```

meltingrabbit, 2023-03-16:
> Cmd_TG_REPEAT(REPLAY)_TLM

FORWARD がただしい，ということなので，動詞はこれで．

meltingrabbit, 2023-04-11:
- 検討はした．実装は https://github.com/ut-issl/c2a-core/issues/537 に引き継ぐ

closed 2023-04-11 by meltingrabbit