ut-issl/c2a-core #255: static_assertを入れるか検討する

meltingrabbit, 2022-02-17:
## 概要
static_assertを入れるか検討する

## 詳細
- https://cpprefjp.github.io/reference/cassert/assert.html
- https://learntutorials.net/ja/c/topic/555/%E3%82%A2%E3%82%B5%E3%83%BC%E3%82%B7%E3%83%A7%E3%83%B3

## close条件
どうなったらcloseできるか

## 備考
なにかあれば

## 注意
- 関連する Projects が存在する場合，それの紐付けを行うこと
- 可能ならば `priority` ラベルを付けること
- 可能ならば Assignees を設定すること
- close するときは結論を明記すること

sksat, 2022-02-17:
まず `static_assert()` と `assert()` は別物です．前者は**コンパイル時**にやる検証で，後者は**実行時**にやる検証．

で，`assert()`はCからある文化ですが(実装としてはダメだったらabortするだけなので)，`static_assert()`はコンパイラが頑張らないといけない．そしてCはコンパイラが頑張れない時代からの言語なので当然そういうのは無い，という．

ただ，C11からは`static_assert()`的な物体である `_Static_assert()` が存在はします．
https://www.ibm.com/docs/ja/zos/2.3.0?topic=declarations-static-assert-declaration-c11

meltingrabbit, 2022-02-17:
あー，完全に誤認してた． < コンパイル時

meltingrabbit, 2022-02-17:
Cだと `#ifdef` で落とすもんね．コンパイル時．

まあプリプロセッサ定義しか使えないんだけど．．．

sksat, 2022-02-17:
前にコンパイル時検査したいな〜〜〜と思ったことはあって，
- `_Static_assert()`をC11以前では無に潰すマクロで包む
- 検査用にC11でビルドを回す

みたいな曲芸をすればできなくはないなあとは思ったんですが，しかし...

sksat, 2022-02-17:
`assert()`は実行時に落とすわけなのでまずどうやって動かすねん/落ちた後どうすんねんという問題があるわけですが，我々にはSILSという武器があるので入れられるところに入れまくるのはかなりアリだと思っています(そしてCIでSILS回したくなってくる)．

meltingrabbit, 2022-02-17:
そうなんよね < SLISがある

もうちょっとで https://github.com/ut-issl/s2e-user-for-c2a-core/pull/7 をマージするので，SILSはととのう．

あとはwingsの公開を待ち，んで，wingsどうCIで動かすか．．．ってのはあるんだけど．．．．

meltingrabbit, 2022-02-17:
ただ，どこで落ちたか，わかりにくいよね．プロセスkillされちゃうだけだと．

sksat, 2022-02-17:
そんなことはなくて，こんなかんじになります．そしてSILSではC2A普通にリンクするので素の`assert()`でいい．

[image]

meltingrabbit, 2022-02-17:
え，行数わかるの，まじか．あれ？いいの？っておもったけど，リリース版は `NDEBUG` で落とすからいいのか．

meltingrabbit, 2022-02-17:
なるほどね

```
#define       assert(x) ( x ? (void) 0 : (fprintf(stderr,\
              "ASSERTION FAILED: "#x" FILE %s,LINE %d \n", \
              __FILE__,__LINE__),abort(),(void)0))
```

sksat, 2022-02-17:
ちょっと例分かりにくかったので差し替えましたが中身も分かる．  
です > リリース版は `NDEBUG` で落とす

sksat, 2023-03-02:
ドライバを書く時にこういう雑な assert もどきを突っ込んで当然便利だったので，こういうかんじのでも c2a-core にあるといいかもという気持ちになってきた
```c
#define assert_eq(_Expression1, _Expression2) { \
	if((_Expression1) != (_Expression2)){ \
		printf(__FILE__ ":%d: assertion error: " #_Expression1 " != " #_Expression2 " = %d \r\n", __LINE__, (_Expression2)); \
	} \
}
```

meltingrabbit, 2023-03-02:
よさそう