ut-issl/c2a-core #275: BC展開時，packet handlerを経由せずに直接packet listをたたいてるのを直したい

meltingrabbit, 2022-03-03:
## 概要
BC展開時，packet handlerを経由せずに直接packet listをたたいてるのを直したい

## 詳細
- https://github.com/ut-issl/c2a-core/blob/03a4b7d403169dc38c9d1a939ce56810cb12873e/Applications/timeline_command_dispatcher.c#L345 でPLを直接叩くのではなく
- https://github.com/ut-issl/c2a-core/blob/03a4b7d403169dc38c9d1a939ce56810cb12873e/TlmCmd/packet_handler.c#L99 に流したい．

AppレイヤでPL直接操作したくないので．

## close条件
直したら

## 備考
- https://github.com/ut-issl/c2a-core/pull/274#issuecomment-1057679196 ででた

yngyu, 2022-03-22:
* https://github.com/ut-issl/c2a-core/issues/296
こういうのにも関連するんですが、これどこに追加して欲しいですか? というのも BC を展開しているのは TL と TaskList の2つあるので...

meltingrabbit, 2022-03-22:
時刻かぶってるのをずらしていれる，は，PHじゃないかなぁ．

で，今のTasklist, cmd disでやってるような，時間過ぎてたらアノマリ発行とか，そういうのは今までどおりの場所でよさそう．

yngyu, 2022-03-22:
じゃあほぼ PL_deploy_block_cmd を PH_deploly_block_cmd で良いですかね?

meltingrabbit, 2022-03-22:
時刻かぶってるのをずらしていれる，は，PLじゃないかなぁ．PL utilでもいいかも．

で，今のTasklist, cmd disでやってるような，時間過ぎてたらアノマリ発行とか，そういうのは今までどおりの場所でよさそう．

BC展開がPLにあるのは違和感あるんだよねぇ（それはBC側で持つべきでは，と）