ut-issl/c2a-core #296: TL (time line cmd) 登録時に，時刻がかぶってると棄却されるので，指定時刻より後で最も近い時刻に登録するラッパーがほしい

meltingrabbit, 2022-03-17:
## 概要
TL (time line cmd) 登録時に，時刻がかぶってると棄却されるので，指定時刻より後で最も近い時刻に登録するラッパーがほしい

## 詳細
ssia

## close条件
できたら

yngyu, 2022-03-30:
前提
* C2A 内の全 Packet は PH_analyze_packet を通って欲しい

があるのを考えると、ずらして入れる場合は  PH_analyaze_packet を何回も呼び出してちょっとずつ入れようとする時刻をずらす感じになるがこれは最悪の時何回もこれを呼び出してTLを走査して、を行うので避けたい  
となると、Packet のフラグにずらして入れるフラグを実装することになるがこれでPacket の仕様変えまくるのも厳しい
個人的には BC の TL上への展開を行うのを PH_analyze_packet 経由で行うのは厳しいので PH_util_deploy_block_cmd みたいなのを作ることになる気がするので PH_util_insert_tl_cmd_asap みたいなのの新設を許して欲しい (結果として前提を破ることになる)

meltingrabbit, 2022-03-31:
> C2A 内の全 Packet は PH_analyze_packet を通って欲しい

を達成するなら

別issueで
```
  CCP_form_tlc();
  PH_analyze_cmd_packet();
```
を
```
CCP_register_tlc
```
にまとめた関数を作る予定で，ここでasapかstrictlyか選べるようにしてasapでは中でPHをよびまくってる（ `PH_analyze_packet ` の返り値をみて，だめだったら，TIをずらして再度 `PH_analyze_packet ` する）のがいいなぁ．（原則論）

ただ，これをすると， `だめだったら，TIをずらして再度` が N回発生するような場合，めちゃくちゃ時間がかかる（N回PLをなめることになるので）のが問題

yngyu, 2022-06-12:
* https://github.com/ut-issl/c2a-core/pull/321
でマージされたので close

meltingrabbit, 2022-06-15:
closeされてなさそうなので，しますね．