ut-issl/c2a-core #338: EH の対応手法として単発 Cmd も欲しい

yngyu, 2022-06-02:
## 概要
EH の対応手法として単発 Cmd も欲しい

## 詳細
現状 EH の対応方法として BC Deploy しかないが、単発 Cmd で十分な時もある。 BC はメモリを食うため、単発 Cmd でも対応してほしい

## close条件
上が実装されたら

## 備考

## 注意

meltingrabbit, 2022-06-05:
単発コマンドを実装するなら，

- Cmdを直接 EH ruleに格納する方法
  - これは，メモリ管理がだるい & 登録がだるい
- BCTのblock idだけでなくposも指定する方法
  - 自由度がたもてるし，よさそう

だが，基本的に，EHは発火後にinactivateされるので，それを戻すためには最低2 cmd必要なはず．．．？ （つまり，この機能，本当にworkするのか？）

meltingrabbit, 2022-07-12:
@seki-hiro 

> EL_ERROR_LEVEL_EHは記録に残らないという認識です。

EL に記録残ってるはずだよ．
`EventLogger.tlogs[EL_ERROR_LEVEL_EH]` などに入ってます．

seki-hiro, 2022-07-12:
```C
/**
 * @brief  TLog のイベントテーブルの上書きをすべて有効化
 * @note   EL_ERROR_LEVEL_EH は基本的にユーザーが操作するものではないので，ここでは有効化されない
 * @param  void
 * @return void
 */
void EL_enable_tlog_overwrite_all(void);
```
これを見て勘違いしていたのですが、`EL_ERROR_LEVEL_EH`のページを見れば記録されているということですね。（上の関数は上書きについての設定で、かつ`EL_enable_tlog_overwrite(EL_ERROR_LEVEL_EH);`により別で上書きもされているため、あんまり関係なかったみたいです）。

ただ運用の時は、`EL_ERROR_LEVEL_HIGH`を一番見る気がするので、ここにきちんと記録されるといいのかなというイメージを持っています。

meltingrabbit, 2022-07-12:
なるほどです．
EHの反応ログは，運用中は結構見てるだろうなぁと思いつつも，

> ただ運用の時は、EL_ERROR_LEVEL_HIGHを一番見る気がするので

はその通りなので，ISSLの次期プロにむけて考えておきます！