ut-issl/c2a-core #454: 名前変更したいもの一覧！

meltingrabbit, 2022-10-06:
## 概要
名前変更したいもの一覧！

## 詳細
ひとまず集めよう

sksat, 2022-10-06:
単体の `Driver` という語を無くし，`IfWrapper` を `Device Driver`, `Driver` を `Component Driver` とするなどの案

meltingrabbit, 2022-10-06:
既存のOS

- https://docs.zephyrproject.org/latest/kernel/drivers/index.html
- https://docs.zephyrproject.org/3.0.0/guides/networking/net-stack-architecture.html

meltingrabbit, 2022-10-06:
- IF Wrapper -> Device Driver: DD
- Driver -> Component Driver: CD
- Driver Super -> Component Driver Super: CDS
- Driver Inctance -> Component Inctance: CI
- IF List: Device Drivre List: DDL

meltingrabbit, 2022-10-06:
device は， SW 的な device の原義に寄せていき，いわゆるコンポは componet として区別していこう（お気持ち表明）

meltingrabbit, 2022-10-06:
`src_user/Drivers/` ，今雑に srx.c とかあるけど，本当は型番とかつけないとだめそうだね． component driver なら（また別で話が進んでる，driver 分離の文脈でも）

meltingrabbit, 2022-10-06:
slack からの引用
```
KOBA789

「コンポドライバは新規で書かないとですね」「でもこのコンポも物理層は UART ですよね」「じゃあデバドラは使いまわせますね」という会話ができる。素晴らしい
```

sksat, 2022-10-06:
現C2A語のIFってIfWrapperにあるやつ全部の抽象化インターフェースではないですよね(たぶん)？Device Listだけは不適当だと思う

meltingrabbit, 2022-10-06:
ですね．正確には Drivrer Super に載せる通信インターフェースです．

フレーム解析やってほしいやつ．

UART, CCSDS, SPI, I2C とか．

meltingrabbit, 2022-10-06:
> Device Listだけは不適当

たしかにな．

meltingrabbit, 2022-10-06:
Communication Interface List とかになってしまうが，，，

sksat, 2022-10-06:
どちらかというと，フレーム解析やってほしい，という特徴の方に即して命名すべきかも

meltingrabbit, 2022-10-06:
てか，これ別に if wrapper にいる必要はなくて， driver super の設定パラメタにしてしまったほうがきれいだな．

sksat, 2022-10-06:
フレーム解析やってほしいタイプのデバイス，とそれを共通化して扱わなければならないためのインターフェース(Cなので関数ポインタテーブル)，そういうインターフェースを使って他のコンポとお話するComponent Driver，という整理にすべきかな

meltingrabbit, 2022-10-06:
マイクロカーネルの用語をこっからツモってきた方がいい説

https://seiya.me/microkernel-book.pdf

sksat, 2022-10-06:
DriverとComponentの違いって何よ，という話があり，「テレコマとして抽象できるもの」をComponentと呼ぶことにして，テレコマ(の型)を引き出す部分をComponent Driverとするのがよさそう，ということに．
あと，`Driver Instance`もかなりC2A語で，これは↑のテレコマの型に実際のコンフィグを注入する太郎であり，かつ真のdriver instanceを配列で持ってたりする(同じものが複数載ってる場合など)ので，Component Serviceとでも呼ぶのがよさそう．

sksat, 2023-03-13:
if_list したい device を StreamDevice とか FrameDevice みたいに名付けて，DriverSuper はそう呼ぶとよさそうと思うなどした