ut-issl/c2a-core #480: コマンド実行時エラーの EL の local_id を改善する

chutaro, 2023-01-24:
## 概要
コマンド実行時エラーの EL の local_id を改善する

## 詳細
CDIS の コマンド実行時エラーの EL で、
- local_id は cdis の関数ptrのアドレス
- note は cmd_id と 実行時status を 16bitずつ分けたやつ

となっているが、分かりにくいうえ、個別で無効化したり EH 対処したりしにくい（特定の Cmd の実行時エラーのみ EL 記録を無効化したい、というようなユースケースが生じた）。ので、local_id を cmd_id、note を実行時ステータスにしたい。（どのCDISで実行されたかは正直あまり重要でない説）

## close条件
変更したら

## 備考
なにかあれば

## 注意
- 関連する Projects が存在する場合，それの紐付けを行うこと
- 可能ならば `priority` ラベルを付けること
- 可能ならば Assignees を設定すること
- close するときは結論を明記すること

meltingrabbit, 2023-02-28:
> 特定の Cmd の実行時エラーのみ EL 記録を無効化したい、というようなユースケースが生じた

- https://github.com/ut-issl/c2a-core/pull/499

を見てて思ったんだけど，EL って group ごとにしか無効化できない気がしてて，どういう感じの想定してます？

@chutaro

chutaro, 2023-03-04:
あれ、そうですね勘違いしてました。。

少なくとも、EHでは対応しやすくなるのかなとは思うのですが。（CDISのアドレス情報っていりますか、、？）

meltingrabbit, 2023-03-09:
- local id を cmd id にする，はかなり賛成
- 今後，衛星運用を自動化していくには，どのエラー化を自動で特定したいので，cdisアドレス情報はnoteにはほしいかも

って感じかなぁ．

meltingrabbit, 2023-03-09:
まあそうすると，localとnoteで情報が足りない．．．となり，

- el group を `CCP_EXEC_STS` ごとにつくる？

というのもあり？
ちょっとだるいか．．．

それか，

- 今 2 つの EL を発行してるが，以下のELをがさっと発行するようにする（ELはgroupごとに無効化できるので，いらないものはuserでとめてもらう？）
  - loral: cmd id, note: cdis
  - loral: cdis, note: err_code
  - loral: cdis, note: exec_sts

うーん，きついな

meltingrabbit, 2023-03-09:
cdisのアドレス持ってるのがうざいので，

https://github.com/ut-issl/c2a-core/blob/aae0e3055ebef4141873ec43724c2a18d9d68c59/TlmCmd/command_dispatcher.h#L29-L37

の cdis の内部状態に idx もたせる？
CDIS_init のなかにstatic counter もたせて，CDIS_init 呼び出し回数の番号を u8 で持たせれば，note に押し込めそう．

meltingrabbit, 2023-03-09:
その場合，local を 上位 8 bit: cdis idx, 真ん中 8bit exec sts，下位 8bit, cmd id とかできるので，

- local id: cmd id だけ
- local id: 上位 8 bit: cdis idx, 真ん中 8bit exec sts，下位 8bit, cmd id （full 情報）

とかできそう？

@chutaro いかが？

meltingrabbit, 2023-03-12:
err_code，u32だし，

１つめ
- local: cmd_id
- note
  - err_code
 
２つめ

- local: cmd_id
- note
  - 上位8bit: cdis idx
  - 次の8bit: exec sts
  - 下位16bit: cmd_id？  <---- ここなに入れよう？ 今冗長だなぁ．．．提案してくれたように err code の 16 bit 入れるのはありな気がしてきた．
 
ともに `EL_ERROR_LEVEL_LOW` （でいいよね？）

とか？（いらなければuserでdisableにしてもらって）

chutaro, 2023-03-13:
- 個人的には EL は一つにまとめたい気持ちですね、、 （err_code で 32bit 使っているものって基本ないですよね？）なので、ISSL側では1つめを disable して使う感じにしようかなと思います
- level low で OK です

^ この方針で良ければこちらで修正します！

meltingrabbit, 2023-03-13:
> err_code で 32bit 使っているものって基本ないですよね？

ISSL の flash の EL は全部 note が読み書きしようとした flash のアドレスなので u32 なきがする．
（他にもアドレス系は u32 かも）

meltingrabbit, 2023-03-13:
> なので、ISSL側では1つめを disable して使う感じにしようかなと思います

良さそう！

初期時にdisableする方法把握してる？

closed 2023-04-11 by chutaro