ut-issl/c2a-core #481: sub OBC のテレメを generate する便利コマンドを追加する

chutaro, 2023-01-24:
## 概要
sub OBC のテレメを generate する便利コマンドを追加する

## 詳細
現状、sub OBC で generate してから MOBC で generate する必要がある。それをセットにしたコマンドがあってもいい。引数は APID と Cmd_id

## close条件
作ったら

## 備考
なにかあれば

## 注意
- 関連する Projects が存在する場合，それの紐付けを行うこと
- 可能ならば `priority` ラベルを付けること
- 可能ならば Assignees を設定すること
- close するときは結論を明記すること

meltingrabbit, 2023-01-24:
- https://github.com/ut-issl/c2a-core/issues/247 のあとにやるぞ！！！（博論．．．．）

chutaro, 2023-01-24:
やりましょう！

meltingrabbit, 2023-01-24:
とはいえ，運用のこと考えると，MOBCへは1Hzで来てほしいけど，GSまで1Hzで来たら困るよね．

1shot用みたいな雰囲気か？

chutaro, 2023-01-24:
1shot を想定していました。今だと毎回2コマンド送る必要があって面倒ですね

meltingrabbit, 2023-01-24:
理解．良さそう

chutaro, 2023-09-25:
メモ
- 2nd obc の TG のコマンドID が必要
- cmd の apid と tlm の apid 両方必要なのを失念していた
- 色々大変なので、実運用も考えて、user側にコマンドを持った方が良いと思った
  - Cmd_TG_generate_AOBC_tlm(AOBCのtlm id)
  - Cmd_TG_generate_XOBC_tlm(XOBCのtlm id)
  - etc

closed 2023-09-25 by chutaro