ut-issl/c2a-core #484: 超 general な OK, NG enum ほしくない？

meltingrabbit, 2023-01-24:
## 概要
超 general な OK, NG enum ほしくない？

## 詳細
雑な OK, NG 返すような関数で，Appごとにenum作るのだるい．
OK, NG, UNKOWN みたいな？

meltingrabbit, 2023-01-24:
@chutaro @hoshika-mas どう思います？

meltingrabbit, 2023-01-24:
@sksat 呼んでおこ

sksat, 2023-01-24:
説はありそう．でも，UNKNOWNが欲しくなるなら個別に enum 切った方がよさそう．気持ちとしては `Result<()>` が欲しいということですよね．

meltingrabbit, 2023-01-24:
> 気持ちとしては Result<()> が欲しいということですよね．

そうそう．

たしかに OK / NG だけのほうがシンプルか

hoshika-mas, 2023-01-24:
本当はこのOK/NGで十分なんだけど仕方なくint（uintですらない）にしているコードをちらほら見かけますね。

chutaro, 2023-01-25:
> 超 general な OK, NG enum ほしくない？

無害なのであって良いと思います。src_user\Library\stdint.h とかで定義する感じですかね？

boolとして使いたい場合は使う、UNKNOWN含め返り値が複雑になりそうな場合は使わない、という方針で良い気がします。

sksat, 2023-01-25:
`stdint.h` は C89 環境のためにあるヘッダなので，作るなら場所は別ですね．result みたいな名前にして個別にヘッダ作っていいと思います．

chutaro, 2023-01-25:
なるほどです。無知な質問であれなんですが、stdbool を使わない理由は環境によっては使えないから、みたいな感じですか？

sksat, 2023-01-25:
ですね．マトモな `bool` 自体が入ったのが C99 からで，user部はともかく c2a-core は C89 環境をサポートする限りは使えないです．
それはそれとして，`bool` とは別に enum を作っておいてもいいと思います．C において enum は結局 `int` なんでコンパイル時の意味的にはあんまり変わりないですが
- 単に `bool` である「値」とは分けて，なんらかの「結果」であるのだ，という明確な意思表示ができる
- 言語的には `bool` と同じでも，外部のスクリプトからなんらかのチェックを自動で走らせられそう

なので enum で作りたいですね．

chutaro, 2023-01-25:
なるほどです。ありがとうございます！