ut-issl/c2a-core #487: MOBC_BL.AOBC_BL.Cmd_hoge ができるようにしたい

chutaro, 2023-01-31:
## 概要
MOBC_BL.AOBC_BL.Cmd_hoge ができるようにしたい

## 詳細
MOBCでBLのキューに入って、TL展開されて、発火したら subOBCのBLとして登録される、というのができると便利。軌道上で、AOBC の BC を書き換える block コマンドを MOBCに持つことができ、1コマンドで色々設定できるようになるため。

現状は、`PL_deploy_block_cmd` で展開するときに、 exec type が TL に書き換えられ、TLが発火して PH で DI_AOBC に送られると `DI_AOBC_dispatch_command` で exec type が RT に書き換えられる。ので、現状はどう頑張っても RT コマンドになってしまう。

実装するには以下が必要に思われる
- exec type に `BL_to_subOBC_BL`, `TL_to_subOBC_BL` みたいなのを追加する
- 引数のTIを2つ指定できるように cmdのSHを拡張する

## close条件
良い方法が見つかったら

## 備考
なにかあれば

## 注意
- 関連する Projects が存在する場合，それの紐付けを行うこと
- 可能ならば `priority` ラベルを付けること
- 可能ならば Assignees を設定すること
- close するときは結論を明記すること

chutaro, 2023-01-31:
書いていてコスト高すぎる気がしてきた。分かりにくくはあるが、Cmd_BC_overwrite みたいなコマンドはあるので、現状はそれを使うで良い気はする

meltingrabbit, 2023-03-20:
2nd obcがリセットしたときなど，2nd obc の BCT が消えた場合，かつ，2nd obc の BCT が不揮発出ない場合は，MOBC から 2nd obc の BCT を再度注入したくなる

chutaro, 2023-04-11:
https://github.com/ut-issl/c2a-core/issues/528 これで解決されそう

meltingrabbit, 2023-05-08:
- https://github.com/ut-issl/c2a-core/issues/547 で解決させる

closed 2023-05-08 by meltingrabbit