ut-issl/c2a-core #553: user 側で実装すべき関数の宣言のかかれたヘッダーを user が持つのではなく， core から提供したい

meltingrabbit, 2023-04-20:
## 概要
user 側で実装すべき関数の宣言のかかれたヘッダーを user が持つのではなく， core から提供したい

## 詳細
例えば，

- https://github.com/ut-issl/c2a-core/blob/afd6ee38c44569298c7fa94945b681ea734680c5/Examples/minimum_user/src/src_user/TlmCmd/user_packet_handler.h#L40

は，core から呼び出される関数で，すべての user が実装している．
こういうものは

- src/src_user/TlmCmd/user_packet_handler.c
- src/src_user/TlmCmd/user_packet_handler.h

と，宣言と定義が .h, .c に分かれて，同じフォルダ（＝ともに user）に存在している．

ヘッダーは core でもちたい？

## close条件
方針決めて実装したら

meltingrabbit, 2023-04-20:
@sksat これ，core に移動するとして，core のどのディレクトリに配置するのがいいんだろうか？

上の例だと， `src/src_core/TlmCmd/user_packet_handler.h` とか？
それとも，こういうたぐい，全部まとめて `src/src_core/include/TlmCmd/user_packet_handler.h` みたいな感じにする？

sksat, 2023-04-20:
`include` みたいなディレクトリは c2a-core にあるすべてのヘッダをそこに移すのでなければ混乱しか生まないのでやめた方がいい

sksat, 2023-04-20:
これこそ真に並列してる概念なんだから並列してる c2a-core のディレクトリにいればよくない？（つまり，`src_core/TlmCmd/` 以下でよくない？）
あんまり何をどう悩んでるかがわかってないかも．

meltingrabbit, 2023-04-20:
なんか，userが実装すべき関数一覧，ぱっとわからないけど，いいのかな，って．

sksat, 2023-04-20:
そもそも "c2a-core" というのはライブラリの単位としては一切正しくなくて，ライブラリとして扱うべきは `src_core/TlmCmd` とかの単位（実運用上そういかないとしても，そのように扱い設計すべき）なので，その中で明らかになっていればよいと思う（そのため，将来的に `src_core/TlmCmd/include` が生える，とかはナシではないかな）．
`src_core/include` みたいなのは "c2a-core" という単位でのリリースをバラしにくくする割には得られるうまみがそこまでないと思う．

sksat, 2023-04-20:
user が実装すべき関数一覧はなんも実装せずに executable build したらリンクエラーで得られるみたいな雑な見方もあるし，なんらかの明示的な指定をするとしても documentation comment とかでよいのではないか（現状の Doxygen でもなんか共通したコメント足すぐらいで一覧は得られるのでは？）