ut-issl/c2a-core #595: pytest のポリシーを考える

meltingrabbit, 2023-05-24:
## 概要
pytest のポリシーを考える

## 詳細
どういう思想，仮定で書くかをかんがえる

## close条件
考えて，ドキュメントにしたら

meltingrabbit, 2023-05-24:
Draft

## 仮定
### スタート時
- 起動直後の状態からスタートする
- 起動直後の状態を仮定するが，その状態であることのアサーションは， 1 pytest の冒頭で行う
  - 例: 電源がOFFであること，など
  - 例: 機能AがStop状態であること，など

### 1 pytest 終了時
- できるだけ pytest 開始時の状態を復元する（がmustでなはい）
- 後続の pytest に悪影響のある状態は困るので，それは必ず復元する

meltingrabbit, 2023-05-24:
FYI @sksat @chutaro pytestのポリシー考えてます．

sksat, 2023-05-24:
一連の pytest が一つの C2A のインスタンスを使い回していること自体をどうにかしたくはありつつ，一旦現状はそうなのでよいと思う

sksat, 2023-05-24:
それはそれとして，依存関係を持ったテストを書きたくなることもあったりはしないんだろうか，というのは気になる．あっても良いとは思っていて，その上である場合は命名などでのなんらかの明示をしたい，みたいな気持ち

meltingrabbit, 2023-05-24:
> 依存関係を持ったテストを書きたくなることもあったりはしないんだろうか

それって 1 pytest にまとめるべき？（多分今そうなってる）

sksat, 2023-05-24:
まあそうか．少なくとも登場人物が1つの SILS な C2A だけならあくまで unit test に徹するべきだな．たぶんあるとうれしいかもしれないのは共通化した処理を切った上で「途中のここで落ちました～」みたいなやつで，別にそれはこまめに assert 書けばいいだけか．